cpetrich/counterfeit_DS18B20 #1: Add code that would help identify fakes

Misiu, 2019-10-21:
Hi,
I just found Your repo, and I must say I'm amazed by work You did.
I think there is one thing missing: a safe code that would allow us to test DS18B20 sensors.
You wrote that You can damage original sensors if You send bad commands, so I want to avoid damaging sensors - I have a couple of original I can use for reference.
Simple ino file that would allow us to read address, compare it to the pattern, read the scratchpad register and check specific bytes, send 0x68 and check if it is 0xff.
That would be really helpful!

cpetrich, 2019-10-22:
Hi Misiu,
thanks. Good point.
I've added discover_fake_DS18B20.ino to the repo that I believe to be safe. It depends on Paul Stoffregen's OneWire library. 

To cover my back discover_fake_DS18B20.ino does not use any commands that are not in the datasheet (i.e. it does not read 0x68 either). It finds all counterfeit sensors I am aware of at the moment.

Hope this helps.

closed 2019-10-25 by cpetrich

Misiu, 2019-10-28:
@cpetrich works perfect :)
I've tested it with sensors from Aliexpress, eBay and genuine. I'm waiting for a couple of [DS18B20-PAR](https://datasheets.maximintegrated.com/en/ds/DS18B20-PAR.pdf) and I will check them with your script. They will be genuine, so I think your script will show the same.

In ino file there is a comment about power-up state and EEPROM:

> It does not test the power-up state and it does not write to or test the EEPROM.

why this isn't tested? Do you think this could damage sensors?

cpetrich, 2019-10-28:
Thanks for your feedback.

The script does is not designed for use in parasitic mode. In particular, it does not force the data line high during temperature conversion. Your mileage may vary. 
The reason I did not design for parasitic mode is that the likelihood of incorrect connection of a standard DS18B20 is high (i.e. leaving VCC floating rather than connecting it to GND).

To test the power-up state, the processor would need to be able to control the power line. Although that could be done easily, a test for chips with internal supercap would need to be added (and that test may be time consuming to run for some chips since the supercap needs to be discharged). The added complexity in hardware and software is currently not necessary to detect fakes, and the script would not be "simple" anymore as per your original request.

The EEPROM is not tested as it appears that such a test would permanently damage chips of Family C: at least the ones I've tested withstood only an extremely small number of write cycles.