cpetrich/counterfeit_DS18B20 #2: UMW DS18B20

N-Storm, 2020-01-11:
Hello,

I've found those chips manufactured by UMW(Youtai Semiconductor Co., Ltd.) and labeled as DS18B20. 

Found here: https://lcsc.com/product-detail/Temperature-Sensors_UMW-Youtai-Semiconductor-Co-Ltd-DS18B20_C376006.html
They also have a datasheet for these. It's in Chinese, but some information could be discovered.

Chips are marked as UMW 18B20 so they are distinguished from Dallas. But we all know how some chips could be re-marked by various fake sellers.

I don't have any of them or ordered any. But probably this information could be useful.

cpetrich, 2020-01-11:
much appreciate the pointer!

I'd dare a prediction based on the datasheet: Family B1. How exciting!

cpetrich, 2020-01-15:
Turns out they are Family B1 indeed.

closed 2020-01-15 by cpetrich

makerhelp, 2021-05-12:
I have bought 10 og these and they self destroy😡