cpetrich/counterfeit_DS18B20 #3: Any idea to which type of fake this one belongs

mdk2412, 2020-02-02:
According to your fantastic sketch:
10-B0-15-16-03-08-00-F1: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  Scratchpad Register: AA/00/B4/B9/FF/FF/0C/10/18
  Info only: Scratchpad bytes 2,3,4 (B4/B9/FF):  not Maxim default values 4B/46/7F.
  Scratchpad byte 5 (0xFF):  ok.
  Scratchpad byte 6 (0x0C):  unexpected value. Error.
  Scratchpad byte 7 (0x10):  ok.
  0x4E modifies alarm registers:  ok.
  0x4E accepts 10 bit resolution:  rejected (want: 0x3F, got: FF). Error.
  0x4E preserves reserved bytes:  ok.
  0x4E accepts 12 bit resolution:  rejected (expected: 0x7F, got: FF). Error.
  0x4E preserves reserved bytes:  ok.
  Checking byte 6 upon temperature change: not necessary. Skipped.
  --> Sensor appears to be counterfeit based on 4 deviations.

Marking reads 1540C4, +726AC. Indentation on the back shows a P.

cpetrich, 2020-02-02:
Hi, thanks for the pointer.

Actually, this is a DS18S20 rather than a DS18B20, so the discussion/code doesn't apply. I cannot comment on the DS18S20 much I'm afraid. 

The code does not explicitly flag wrong family codes as there are DS18B20 counterfeits out there that let you overwrite it.

mdk2412, 2020-02-04:
Hi,

thank you for this info! I was about to through this sensor away. I wondered why a sensor marked with P would be fake. As it is apparently not I will find use for it. 

BR

closed 2020-02-08 by cpetrich