cpetrich/counterfeit_DS18B20 #6: List of vendors with counterfeit and original

JsBergbau, 2020-03-27:
Hi cpetrich,

thanks for providing all this info and the fakecheck script.
I wonder if it is possible to provide a list where you bought counterfeit and where original sensors. This helps choosing the right store.

cpetrich, 2020-03-27:
Hi,
fair enough, I'll look into compiling a list of sources. 

If you want to buy original sensors get then from the Maxim web store or from one of the authorized distributors in your area listed here: https://www.maximintegrated.com/en/aboutus/contact-us/distributor-offices.html
If you just need 4 sensors for your project then Maxim will gladly courier them to you for free (select the Order tab on their DS18B20 page and click on Sample). If $15 for shipping is within your budget then get them from an official distributor.

I'm not going to vouch for anybody else based on two anecdotes: 
1. In one purchase of authentic sensors from a reputable but non-authorized distributor I had 3 sensors that gave me an offset of 0.88 C at the freezing point. It is beyond me how that can happen but I did repeat those measurements many times: it was not a measurement error, and all of those bad guys had an almost identical (way too high) offset. 
2. In another case I ordered probes with identical part numbers from the same retailer twice. The first delivery contained an authentic sensor, the second contained a clone. Same part number, ordered a month or two apart.

If you want to buy decent waterproof probes that contain original sensors you'll probably have to buy from a store that caters to tinkerers. Maybe you can ask their tech people for the ROM code of a recent sensor. You'll probably face a price tag somewhere above $7 a piece, more like $10. I've had people tell me that their probes contain original sensors only to find that they contained Family B clones. I've paid $15 for one probe that contained a Family D2 sensor -- that store refunded the full costs including shipping though after I told them that parasitic power doesn't work. Faced with this mess our solution was to have a 4 digit-number of them custom produced. If like $10 a piece plus shipping and customs import fees to your country are not an issue then I can pack up some for you. Otherwise I refer to anecdote 2 above for why I won't drop names and the first three sentences of this paragraph for how one could proceed.

cpetrich, 2020-04-04:
added a list of sources

closed 2020-04-04 by cpetrich

MatthiasSchinzel, 2020-09-11:
I bought a ds18b20 from www.develektro.com, which is a shop in germany where private persons can buy farnell stuff. Apparently they (develektro and farnell) share the same warehouse.  This seems to be the only source of guaranteed original sensors in germany to my knowledge. Therefore it is maybe worth it to add this one to the list too. I of course also ran your check without any problems.

Erriez, 2020-09-12:
@MatthiasSchinzel Two weeks ago I've purchased DS18B20 at https://reichelt.nl = https://reichelt.de, my favorite electronics webshop in Germany. These sensors are genuine and works reliable at low voltage and accurate temperature.

MatthiasSchinzel, 2020-09-12:
Yes, sure there are many suppliers of electronics in Germany. However maxim is not mentioning one as certified supplier, including reichelt.

cpetrich, 2020-09-12:
@MatthiasSchinzel This sounds intriguing and shows how blurry the line can get. May be a case for Maxim customer support to clarify.

Although it may not be a solution for everybody, an alternative route could be to check whether other authorized distributors ship to private customers in Germany for free and without customs charges if the order is big enough (I believe Mouser, for example, does as of writing).