cpetrich/counterfeit_DS18B20 #7: Counterfeit sensors vanishing after some time

JsBergbau, 2020-04-10:
Hi cpetrich,

thanks for all your work. I've ordered some waterproof sensors from china and all this batch disappears after some time on RaspberryPI, the AZDerlivery counterfeits on the same bus stay present. The china counterfeits can only be made working again by disconnecting and connecting. Don't know if thats interesting for you. Did you make some longterm tests and seeing a simiar behaviours with some sensors? I could also send them to you for free, if you're interested.

cpetrich, 2020-04-10:
Thanks,

what's the ROM of the sensors in the probes, please?
Disappears while immersed in water, or even in air?

As for sensors:
We had problems operating multiple probes with Family D1 sensors on the same bus, limit was somewhere in the low single digits, forgot where exactly. That problem arose immediately. This was the first batch of probes we tested, and since then we've called probes with Family D1 and D2 sensors a loss, treated them as garbage, and haven't looked closer. 

As for probes:
We had long term problems after immersion in saltwater, sometimes arising within one day, sometimes within 2 weeks, but often they would still work after 2 months. This was due to leaky "waterproofing" of the probes rather than the sensors.

I've also seen probes with a poor solder job on the Vcc line.

JsBergbau, 2020-04-10:
[image]

> Disappears while immersed in water, or even in air?

Even in air. 10 sensors were on the bus and one after another was polled in a bash loop with ```cat /sys/bus/w1/devices/$address/w1_slave```
After some time they disappeared and needed disconnect and reconnect. When 4 AZDelivery counterfeits were connected with 6 counterfeits these 4 sensors were still present and reportet temperature.

cpetrich, 2020-04-11:
Interesting, that's Family D2 causing issues.

Looks like the Family D1/D2 manufacturer is anonymous for good reason.

closed 2020-04-11 by cpetrich