cpetrich/counterfeit_DS18B20 #9: Temperature offset

cassiel74, 2020-05-12:
I have those two guys and it seems to me that they've a temperature offset like in A2 family but on the higher ones. Eg ~3.5° at around 45° while aroung ~0° at 18°
What do you think about? 

```
Searching the 1-Wire LAN
28E7761E5F1401C3 : DS18B20 Temperature Sensor
ROM #0 : 28E7761E5F1401C3

Searching the 1-Wire LAN
28A248805A1401BA : DS18B20 Temperature Sensor
ROM #0 : 28A248805A1401BA

```

cpetrich, 2020-05-14:
Interesting. 

1. I have tested temperature readings at temperatures other than 0 C sporadically, only, because that takes considerably more effort to get right than a measurement in an ice--water bath at 0C. I don't expect the offsets at 0 C will tell you much about the errors at other temperatures, though.

2. I have two probes with sensors that have ROMs similar to yours, and those are Family B1:
28-96-E7-8E-5A-14-01-C2
28-02-B7-13-5F-14-01-73

cassiel74, 2020-05-14:
Ok, so is not that easy to tell which family they belongs to.
I see that A2, D1 and D2 have offsets up to 3° so my guess is that mine should be in those families.

For the sake of clarity, I am not really doing tests but just noticed this discrepancy between my heat pump send temperature and the DS18B20 being in contact on the send/return pipes [similar to this](https://www.google.com/url?sa=i&url=https%3A%2F%2Fwww.itsensor.it%2Fsonda-di-temperatura-a-contatto-per-tubazione-con-uscita-passiva-cod-altf-1.html&psig=AOvVaw0b6IEb2KieBLJ_bSqtZWak&ust=1589533574379000&source=images&cd=vfe&ved=0CAIQjRxqFwoTCJiZsZiAs-kCFQAAAAAdAAAAABAE)

cpetrich, 2020-05-14:
Make sure there's insulation surrounding the pipe with the sensor otherwise your measurement error will increase the more the pipe temperature differs from the ambient temperature (i.e. you measure a temperature that lies between pipe and air temperature).

cassiel74, 2020-05-14:
Yes, there's insulation surrounding the pipe and thermal paste all around the probe.

cpetrich, 2020-05-24:
For future reference, I've added a sketch that determines what Family a sensor belongs to based on low-level implementation details (i.e. using undocumented function codes): ``classify_fake_DS18B20.ino``

closed 2020-05-24 by cpetrich

cassiel74, 2020-05-24:
sketch_may24a.ino: In function ‘void loop()’:
sketch_may24a.ino:391:1: error: jump to label ‘err_C’ [-fpermissive]
sketch_may24a.ino:380:68: note:   from here
sketch_may24a.ino:381:15: note:   crosses initialization of ‘uint8_t cfg2’
sketch_may24a.ino:391:1: error: jump to label ‘err_C’ [-fpermissive]
sketch_may24a.ino:371:68: note:   from here
sketch_may24a.ino:381:15: note:   crosses initialization of ‘uint8_t cfg2’
sketch_may24a.ino:372:15: note:   crosses initialization of ‘uint8_t cfg1

cpetrich, 2020-05-24:
changed the script so it compiles without the (Arduino IDE default) -fpermissive compiler option

cassiel74, 2020-05-25:
it compiles but I got this output (I am running it on linux IDE)

```
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x20
```

cpetrich, 2020-05-25:
1. avrdude is the program used to upload the program to the board, i.e. this error has nothing to do with the script and should appear regardless of what program you try to upload. Consider checking the Arduino forum for pointers.

2. I've never used the Linux IDE so here are some things to check based on the way the Windows version works. I'm assuming you connect the board to the computer through a plain USB cable and you use something like an Uno or so that doesn't need the reset button pushed manually.
    - select the board in "Tools/Board"
    - select the port the board is connected to in "Tools/Port"
    - close any terminal (program) connected to that port (also check any other instances for the IDE)
    - use "Sketch/Upload" rather than "Sketch/Upload using programmer"