cpetrich/counterfeit_DS18B20 #11: Many claims rely on unpublished investigations

schappim, 2020-07-13:
As a part owner of a company that distributes products that purport to contain the DS18B20 chip we find the claims concerning. Most likely we too have been duped by our vendors on the legitimacy of these products.
Thank you for taking the time to investigate this.

[A commentor on Hacker News does however make a good point](https://news.ycombinator.com/item?id=23816170):

>I am a bit disappointed about that a lot of his claims how to identify clones are referenced with [5] own research. I mean doesnt one need confirmation from the IP holder?

You cite "Own investigations 2019, unpublished." 107 times. 

Could you please share these or get confirmation from the manufacturer?

Thanks again for your work.

cpetrich, 2020-07-13:
Good point!

First off, "Own investigations 2019, unpublished." is the whole point of the repository. If I had found that information somewhere rest assured I wouldn't have bothered investigating myself and trying to put some systematic to it. If I had one wish for the future it would be that everybody posting about issues with DS18B20 sensors stated their ROM.

1. Confirmation from the manufacturer: I believe it would carry more weight if somebody with actual commercial interest in the specifics got in touch with them. Maybe Maxim can put up relevant information on their FAQ page. This work started since I wanted to know whether probes I had deployed contained clones or genuine parts. I am able to tell now at my required level of confidence.

2. As you know, documenting something takes more time than doing it. To ensure efficient use of time, which of the 107 statements or conclusions would you like me to start with?

cpetrich, 2020-07-25:
I started to add some details of the investigation with the intention to (very slowly) add more. Leave a comment if you would like me to prioritize something.

closed 2020-07-25 by cpetrich

theoroborus, 2024-11-18:
As a scientist who plan to use DS18B20 probes, your work is valuable. It should be published and I'll do not miss the opportunity to quote your work even if it's unpublished.