cpetrich/counterfeit_DS18B20 #13: Clarification on "Obtained neither probes nor chips in 2019"

sethtroisi, 2020-07-13:
From [Obsolete as of 2019](https://github.com/cpetrich/counterfeit_DS18B20#obsolete-as-of-2019)

> Obsolete as of 2019
>Obtained neither probes nor chips in 2019

>ROM patterns [5,7]: 28-tt-tt-ss-00-00-80-crc
>Example ROM: 28-9E-9C-1F -00-00-80- 04

I purchased several [probe from Amazon](https://www.amazon.com/gp/product/B00CHEZ250) in May 2016,
When I run `classify_fake_DS18B20.ino`
I get 

    28-28-2D-28-00-00-80-FD: (Could not identify Family.)

This doesn't quite match the obsolete pattern or the good pattern; I'm assuming this means I have a counterfeit probe but not one that you've seen?

cpetrich, 2020-07-13:
Yes, your -00-00-80-'s are counterfeit, and yes, I wasn't able to purchase those last year no matter how hard I tried: I always ended up with (probes containing) Family D2 sensors.

If you have spare probes with -00-00-80- sensors that you are happy to send me then I'll have a closer look to see where they fit in -- may be of historical interest and could give context for what we're dealing with today.

sethtroisi, 2020-07-15:
@cpetrich I found a number of extras. If you send me an address (you can email me at sethtroisi@google.com) I'll post them to you.

cpetrich, 2020-07-15:
Much appreciated!

closed 2020-08-15 by cpetrich