cpetrich/counterfeit_DS18B20 #15: DS18B20 Waterproof Temperature SONOFF sensor

DSgUY, 2020-08-15:
Have you test this waterproof sensors? I'm having some differences between 2 similar sensor and to far from a digital or mercury thermometer I used as reference.

cpetrich, 2020-08-15:
I haven't, actually. But it would be a good one to look into.

DSgUY, 2020-08-17:
I'm getting **Family D2 (Clone w/o parasitic power)** from those waterproof sensor? What that this mean? There are not realiable? I want to log temperatures around 5 celcius from inside a refrigerator and I need to be a realiable measure.

cpetrich, 2020-08-17:
Please see the plot I added to the page just now that shows the temperature errors of sensors we measured at 0 C. The sensor you mention is in the D2 (97) group. Also, we had a few sensors with ROM patterns 28-xx-xx-94-97-0E-03-xx, one of which read +1 C in an ice-water bath.

My interpretation of this plot is that readings of your sensors could be anywhere from spot-on to wrong by as much as a degree or two at 0 C.