cpetrich/counterfeit_DS18B20 #18: raspberry pi version

pabloandresm, 2020-08-24:
Very nice investigation and code.

It is a shame that there is no C or Python code for Raspberry Pi. If yould be a very nice too have.

thanks

cpetrich, 2020-08-25:
ok, point taken...

hurra, 2021-04-16:
You propably can check the serial number with

    hexdump -v -e '7/1 "%02x-" 1/1 "%02x" "\n"' /sys/devices/w1_bus_master1/<family-id>/id

scruss, 2022-02-16:
And this in MicroPython for the Raspberry Pi Pico/RP2040, assuming pin 16 as above:

    # print hex IDs of all ds18x20 sensors found on a pin
    from machine import Pin
    from onewire import OneWire
    from ds18x20 import DS18X20
    
    ds = DS18X20(OneWire(Pin(16)))
    
    for i, sensor in enumerate(ds.scan()):
        print(i, ":", '-'.join(["%02x" % j for j in sensor]))

Other implementations might use something like `Pin('PE0')` to define a pin.

cpetrich, 2024-03-18:
I've copied a Python code fragment in issue https://github.com/cpetrich/counterfeit_DS18B20/issues/20#issuecomment-2005291905
That could be used as a basis for porting the arduino code to Python.