cpetrich/counterfeit_DS18B20 #19: Please check this tip, it might be interesting to update the library.

rtek1000, 2020-10-18:
`The DS18X20 B7 chip die (found in the DS18B20, DS1820S, and DS1822 products) can experience EEPROM data corruption failures during power on reset.`

`Since the EEPROM holds internal trim values (in addition to the user data in the TH and TL registers) that control the conversion process of the DS18X20 this may show up as inaccuracy of temperature readings. And can cause temperature measurement errors of up to ±60°C.`

Source: [[DS18B20, DS18S20, and DS1822] Potential corruption issues!?](https://forum.arduino.cc/index.php?topic=709615.0)

cpetrich, 2020-10-18:
Fair enough, that point should probably be made explicitly. I've added a pointer in the Obsolete list as that seems to be the best way to stay within the scope of the page: *what's on the market as of 2019*. I tried to buy old chips but didn't get anything older than 2009, and all of the chips I got had C4 dies.

The B7 die issue is documented in a somewhat obscure Maxim application note "DS18x20 EEPROM Corruption Issue". That application note is mentioned in the man pages of the OWFS project. The OWFS project implemented a solution to deal with the the B7 bug and least the B7 and C2 dies were mentioned on the OWFS mailing list.

The best thing I could think of doing is to emit a generic warning if the ROM is associated with a pre-C4 die. I don't know where to set the threshold, though. For reference, the earliest ROM I have (which happens to belong to a C4 die) is 28-3C-16-5D-02-00-00-64 with date code ``0939C4``.

rtek1000, 2020-10-21:
> 1. Read the 18X20's ROM code. Look at only the serial number portion of the ROM code: 0xCC00000SSSSSSSFF; where CC is the CRC; 00000 is the customer code; SSSSSSS is the serial number; FF is the family code

Source: [https://dutta.csc.ncsu.edu/iot_spring17/wrap/1-Wire-Design%20Guide%20v1.0.pdf](https://dutta.csc.ncsu.edu/iot_spring17/wrap/1-Wire-Design%20Guide%20v1.0.pdf) Pg. 71.

> 28-3C-16-5D-02-00-00-64

0xCC-00000-SSSSSSS-FF 
0x64-00000-**25D163C**-28

> DS1822
> Starting ROM code of rev B7            4A-00000-008978A-22
> Ending ROM code of rev B7              3B-00000-00CB819-22
> Starting ROM code of rev C2             62-00000-00CB81A-22
> 
> DS18S20
> Starting ROM code of rev B7             E0-00080-0591D20-10
> Ending ROM code of rev B7             AB-00080-080885F-10
> Starting ROM code of rev C2            62-00080-0808860-10
> 
> DS18B20
> Starting ROM code of rev B7           06-00000-0545010-28
> Ending ROM code of revB7             5E-00000-0662B4F-28
> Starting ROM code of rev C2          21-00000-0662B50-28

closed 2024-03-18 by cpetrich