cpetrich/counterfeit_DS18B20 #21: P marking inside the indent on the rear of the case  is potentially not 100% "proof" of genuine sensor.

WO15, 2020-10-28:
ordered several DS18B20s (around October 25) from Farnell.com (official distributor in EMEA)
https://www.maximintegrated.com/en/aboutus/contact-us/distributor-offices.html

Farnell has a PCN Product Change Notice from Maxim on their site
http://www.farnell.com/datasheets/3094172.pdf
EXPECTED PCN SHIP DATE:April 14, 2020
CHANGE FROM: -Maxim commercial products in TO92 package manufactured at current subcontractor CIRTEK/PHILIPPINES
CHANGE TO: -Additional Assembler Greatek in Taiwan/R.O.C

If the P marking inside the indent on the rear of the case is indeed indicating that the sensors are assembled/manufactured in Philippines, it seems that after April 2020 (at least part of) the official sensors, produced in Taiwan/R.O.C. will not have an P marking. Probably wise to indicated this in your explanation.

BTW sensors I received did have an P, full details below.
serial                     ; marking
28-00000cd7e???;  DALLAS, 18B20, 2031C4,+882AB; P in indent on back

WO15, 2020-10-28:
Noticed the product change notice does not contain a typenumber, hereby the source url on the Farnell webpage
https://uk.farnell.com/maxim-integrated-products/ds18b20/temperature-sensor/dp/2515553
under technical docs Product Change Notice EN , the XLS file they refer to in the PDF seems not available on the Farnell website.

cpetrich, 2020-10-29:
Much appreciate the pointer!!

It applies to the DS18B20+ indeed (and DS18B20+PAR, DS18B20+T&R, DS18B20-SL+T&R) according to a copy of that Process Change Notice with attachments on futureelectronics.com.

I've edited the page to reflect the possibility that indent marks beyond ``P`` will appear from 2020 onward. Will be interesting to get a hold of some of those.

etas-github-sb, 2020-12-21:
Ordered 2 DS18B20+ and 2 DS18B20+PAR directly from Maxim. Both were in the same parcel originating from Manila, Philippines. However, the DS18B20+PAR did _not_ have the P in the indent, while the DS18B20+ had. Both were manufactured in 2020.

cpetrich, 2020-12-21:
interesting. Although the +PAR variant is out of scope of the analysis (as noted further down on the page, they seem to have run through a different calibration line since at least 2016, for example, and I've only ever looked at a handful of them), what is the country of origin of the DS18B20+PAR according to the packing slip or the anti-static package the sensors were in?

etas-github-sb, 2020-12-22:
As stated in https://github.com/cpetrich/counterfeit_DS18B20/issues/22#issue-772414488 (sorry for mixing threads), the two non-P-indented chips have a country of origin "TW", which matches to the note mentioned above.

closed 2024-03-18 by cpetrich