cpetrich/counterfeit_DS18B20 #22: ROM valid for identification of fakes?

etas-github-sb, 2020-12-21:
Thanks for sharing all this information with the web. I came across this after having bought 5 DS18B20 on ebay Germany, which I couldn't get to work in 1-Wire, i.e. bus-powered, mode. It took me some hours of debugging before looking for fake chips.

Afterwards, I got 2 DS18B20+ and 2 DS18B20+PAR directly from Maxim. They work smoothly in bus-powered mode with several meters of bell wire between all of them. So, I presume they are "good" and original.

However, I think they do not match the ROM pattern you found out. Having no Arduino available currently, I cannot run your software, but here is the info that I got from my Raspberry:

```
pi@raspberrypi:~ $ cat /sys/bus/w1/devices/
28-00000d1efbf6/ 28-00000d1efed4/ 28-00000d2b5549/ 28-00000d2b7e32/ w1_bus_master1/  
pi@raspberrypi:~ $ cat /sys/bus/w1/devices/28-00000d1efbf6/w1_slave 
f4 00 4b 46 7f ff 0c 10 fa : crc=fa YES
f4 00 4b 46 7f ff 0c 10 fa t=15250
pi@raspberrypi:~ $ cat /sys/bus/w1/devices/28-00000d1efed4/w1_slave 
f3 00 4b 46 7f ff 0d 10 ee : crc=ee YES
f3 00 4b 46 7f ff 0d 10 ee t=15187
pi@raspberrypi:~ $ cat /sys/bus/w1/devices/28-00000d2b5549/w1_slave 
f6 00 4b 46 7f ff 0a 10 d6 : crc=d6 YES
f6 00 4b 46 7f ff 0a 10 d6 t=15375
pi@raspberrypi:~ $ cat /sys/bus/w1/devices/28-00000d2b7e32/w1_slave 
f6 00 4b 46 7f ff 0a 10 d6 : crc=d6 YES
f6 00 4b 46 7f ff 0a 10 d6 t=15375
```
You can see the "ID" as well as the scratchpad content. Both does not match exactly your findings. 

Date code is 2041 and 2037, respectively. Origin is "PH" for the DS18B20+ and "**TW**" for the DS18B20+PAR. And indeed, the DS18B20+PAR do not have the P mark in the indent on the back.

N-Storm, 2020-12-21:
DS18B20+ might be slightly different. I guess this project investigated only DS18B20 originals, without the "plus". Thus, the + version might follow a different ROM pattern even it's shares the same datasheet.

cpetrich, 2020-12-21:
@etas-github-sb, 1. thank you for sharing your observations on the "new" TW-origin chips. 
2. Your dumps are actually consistent with the description of Family A1.
- The scratchpad register content is compatible with Family A after a temperature conversion at around 15 °C. What you show is not the content at power-up. However, you see how the byte following ff (``<byte 6>``) depends on the value of the first byte (``<byte 0>``), which is a distinguishing behavior of Family A (the lower nibble of ``<byte 0>`` and ``<byte 6>`` add up to 0x10, e.g. in your first example: 0x4 + 0x0c = 0x10).
- It is unfortunate that there are two conventions in the wild on how to write the 1-wire ROMs. What you write as ``28-00000d1efbf6`` I write as ``28-FB-1E-0D-00-00-F6``. I.e. I write the ROM as a sequence of bytes the way they are reported by the chip itself while your way is based on an interpretation of the ROM and written as ``<Family Code>-<48-bit Serial Number><CRC>``. Your ROMs are consistent with Family A1.

@N-Storm, the "+" indicates that the sensors are RoHS, and the DS18B20 have been RoHS for ages, certainly for all ``C4`` dies, i.e. as far back as 2009. In other words, the analysis of the state as of 2019 is actually about DS18B20+. Will edit for clarity, thanks!

etas-github-sb, 2020-12-22:
Thanks for this clarification. I didn't realize that the ROM content was just mangled around by the w1 module of Raspian. Then at least your investigation is confirmed by newer samples. :) The PAR models seem to follow the same rule then.

This "issue" can be closed then.

closed 2020-12-22 by etas-github-sb