cpetrich/counterfeit_DS18B20 #29: arduino library

koendv, 2022-08-24:
Hi. Thank you for your original research about the DS18B20 temperature sensor.
I'd like to submit the following for your consideration: your software, repackaged as an Arduino library. 

[ClassifyDS18B20.zip](https://github.com/cpetrich/counterfeit_DS18B20/files/9413989/ClassifyDS18B20.zip)

If your project uses a DB18B20, this library may help you if suddenly a batch of boards acts strange.

cpetrich, 2022-08-24:
Thank you for your effort!

Since I won't have time to maintain this as a library in the Arduino ecosystem you would need to find a way to clarify that you are the maintainer of the submitted library dealing with issues that may arise. I have no experience in how this should be done best.

Besides, the code is mildly out of date since Family E (new on the page as of 2022) is not explicitly accounted for.

koendv, 2022-08-25:
The NS18B20 is not in classify_fake_DS18B20.ino yet.

If you do the NS18B20, I'll do the library. OK?

koendv, 2022-09-03:
OK, done. [library](https://github.com/koendv/CheckDS18B20)

closed 2022-09-03 by koendv

cpetrich, 2022-09-03:
Thank you for the notice.
Please read the license, specifically point 4b, and edit the source files accordingly.

koendv, 2022-09-05:
OK. Like this?

koen

 *   File:    classify_fake_DS18B20.ino
 *   Author:  Chris Petrich
 *   Version: 25 May 2020
 *
 *   Source:  https://github.com/cpetrich/counterfeit_DS18B20/
 *   Documentation:  https://github.com/cpetrich/counterfeit_DS18B20/
 *
 * Arduino library:
 *
 *   Author:  Koen De Vleeschauwer
 *   Source:  https://github.com/koendv/CheckDS18B20