cpetrich/counterfeit_DS18B20 #38: How to get the relevant conversion time.

mayjack0312, 2024-06-20:
Very good project, but I want to ask outside the line: How to modify the code to obtain the conversion time of each DS18B20 and print it out?

mayjack0312, 2024-06-20:
For example, I want to get the corresponding times in the table below one by one.
[image]

lalo-uy, 2024-06-20:
You send the start conversion cmd and time until the bus line goes low.

mayjack0312, 2024-06-20:
> […](#)

Can you suggest a sample code? The code conversion times I wrote before were not quite correct. I almost cried from torture o(╥﹏╥)o

uzi18, 2024-06-20:
Use this function:
_https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/65112b562fd37af68ed113c9a3925c09c4529e14/DallasTemperature.cpp#L411_

mayjack0312, 2024-06-20:
> Use this function: _https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/65112b562fd37af68ed113c9a3925c09c4529e14/DallasTemperature.cpp#L411_

This is the code I wrote yesterday. Although I found that there are many potential bugs in the writing, I still don’t know how to modify it.
```
#include <OneWire.h>
#include <DallasTemperature.h>

#define ONE_WIRE_BUS 2

OneWire oneWire(ONE_WIRE_BUS);

DallasTemperature sensors(&oneWire);

DeviceAddress DS18B20_ID;

void setup() {
  Serial.begin(9600);

  unsigned long initStartTime = micros();

  sensors.begin();

  unsigned long initEndTime = micros();

  Serial.print("Initialization Time: ");
  Serial.print(initEndTime - initStartTime);
  Serial.println(" us");
}

void loop() {
  if (Serial.available() > 0) {
    String command = Serial.readStringUntil('\n');
    command.trim();

    if (command == "GET") {
      unsigned long waitStartTime = micros();

      delayMicroseconds(550);

      unsigned long waitEndTime = micros();

      unsigned long convStartTime = micros();

      sensors.requestTemperatures();

      unsigned long convEndTime = micros();

      unsigned long readCmdStartTime = micros();

      float temperature = sensors.getTempCByIndex(0);

      unsigned long readCmdEndTime = micros();

      unsigned long readRegStartTime = micros();

      Serial.print("Temperature: ");
      Serial.print(temperature);
      Serial.println(" C");

      unsigned long readRegEndTime = micros();

      Serial.print("Wait for Bus Release Time: ");
      Serial.print(waitEndTime - waitStartTime);
      Serial.println(" us");

      Serial.print("Trigger Conversion Time: ");
      Serial.print(convEndTime - convStartTime);
      Serial.println(" us");

      Serial.print("Send Read Command Time: ");
      Serial.print(readCmdEndTime - readCmdStartTime);
      Serial.println(" us");

      Serial.print("Read Register Value Time: ");
      Serial.print(readRegEndTime - readRegStartTime);
      Serial.println(" us");
    } else {
      Serial.println("Unknown command.");
    }
  }
}
```

lalo-uy, 2024-06-20:
You don’t know how requestTemperatures() operates. Could just wait 1s to be
sure.
Get the onewire ds18b20 example, and after request the temp, instead
delay(800) do a loop doing digitalRead() of the bus til is low.

mayjack0312, 2024-06-20:
> You don’t know how requestTemperatures() operates. Could just wait 1s to be sure. Get the onewire ds18b20 example, and after request the temp, instead delay(800) do a loop doing digitalRead() of the bus til is low. El El jue, 20 jun. 2024 a la(s) 13:48, LMY ***@***.***> escribió:
> […](#)

Please help me see if this code is written appropriately. Thank you.
```
// 新功能：请求温度后，延迟800毫秒，循环执行总线的digitalRead()直到低电平，并获取之间的时间并返回
sensors.requestTemperatures();  // 请求温度
delay(800);  // 延迟800毫秒

unsigned long startTime = millis();  // 记录开始时间
while (digitalRead(pin_onewire) != LOW) {
  // 循环直到低电平
}
unsigned long endTime = millis();  // 记录结束时间

unsigned long duration = endTime - startTime;  // 计算时间差
Comm.print(F(" 低电平检测时间: "));
Comm.print(duration);
Comm.println(F(" 毫秒"));
```

mayjack0312, 2024-06-20:
> You don’t know how requestTemperatures() operates. Could just wait 1s to be sure. Get the onewire ds18b20 example, and after request the temp, instead delay(800) do a loop doing digitalRead() of the bus til is low. El El jue, 20 jun. 2024 a la(s) 13:48, LMY ***@***.***> escribió:
> […](#)

Please help me see if this code is written appropriately. Thank you.
```
sensors.requestTemperatures();
delay(800);

unsigned long startTime = millis();
while (digitalRead(pin_onewire) != LOW) {
}
unsigned long endTime = millis();

unsigned long duration = endTime - startTime;
Comm.print(F(" Low level detection time: "));
Comm.print(duration);
Comm.println(F(" ms"));
```

uzi18, 2024-06-20:
```

sensors.requestTemperatures();

unsigned long startTime = millis();
while (!sensors.isConversionComplete()) {
}
unsigned long endTime = millis();

unsigned long duration = endTime - startTime;
Comm.print(F(" Low level detection time: "));
Comm.print(duration);
Comm.println(F(" ms"));
```

mayjack0312, 2024-06-21:
Through a new attempt, I discovered a rather difficult problem. When trying to measure the temperature conversion time, a manual delay must be added between the initialization command and the conversion command. However, this leads to errors in judgment or inaccurate delays. I would like to seek help on how to solve this problem. I tried monitoring the high and low signals of the DQ line, but it seems that the pins of the Arduino Uno do not support this.

uzi18, 2024-06-22:
show complete code and point where is your problem ... please show also output from Serial Monitor

mayjack0312, 2024-06-22:
> show complete code and point where is your problem ... please show also output from Serial Monitor

I tried several methods but none of them worked very well
[code.zip](https://github.com/user-attachments/files/15936323/code.zip)
[image]
[image]
[image]
[image]

The only time it succeeded was to display the time it took to read the temperature.
[image]

uzi18, 2024-06-22:
Please paste actual code to review

mayjack0312, 2024-06-22:
> Please paste actual code to review

<details>
<summary>1</summary>

```
#include <OneWire.h>
#include <DallasTemperature.h>

#define ONE_WIRE_BUS 2

OneWire oneWire(ONE_WIRE_BUS);

DallasTemperature sensors(&oneWire);

DeviceAddress DS18B20_ID;

void setup() {
  Serial.begin(9600);

  unsigned long initStartTime = micros();

  sensors.begin();

  unsigned long initEndTime = micros();

  Serial.print("Initialization Time: ");
  Serial.print(initEndTime - initStartTime);
  Serial.println(" us");
}

void loop() {
  if (Serial.available() > 0) {
    String command = Serial.readStringUntil('\n');
    command.trim();

    if (command == "GET") {
      // 刷新序列
      unsigned long refreshStartTime = micros();
      sensors.begin();
      unsigned long refreshEndTime = micros();

      unsigned long waitStartTime = micros();

      delayMicroseconds(550);

      unsigned long waitEndTime = micros();

      unsigned long convStartTime = micros();

      sensors.requestTemperatures();

      unsigned long convEndTime = micros();

      unsigned long readCmdStartTime = micros();

      float temperature = sensors.getTempCByIndex(0);

      unsigned long readCmdEndTime = micros();

      unsigned long readRegStartTime = micros();

      // 所有测量完成后再打印结果
      unsigned long readRegEndTime = micros(); // 声明并赋值 readRegEndTime

      Serial.print("Temperature: ");
      Serial.print(temperature);
      Serial.println(" C");

      Serial.print("Refresh Sequence Time: ");
[... 383 lines cut ...]
  ds18b20.negative_flag = neg_flag;
  ds18b20.temperature = temperature;
}

void DS18B20_Test() {
  DS18B20_GetTemperature();
  delay(1000);
}

void setup() {
  // 初始化串口
  Serial.begin(9600);
  pinMode(TIME_MONITOR_PIN, INPUT);
}

void loop() {
  DS18B20_Test();
  Serial.print("Temperature: ");
  Serial.print(ds18b20.temperature);
  Serial.println(" C");
  while (1); // 停止循环，等待reset
} 
```

</details>

lalo-uy, 2024-06-22:
El El sáb, 22 jun. 2024 a la(s) 11:27, LMY ***@***.***>
escribió:

> #include <OneWire.h>
>
>
> // OneWire DS18S20, DS18B20, DS1822 Temperature Example
> //
> // http://www.pjrc.com/teensy/td_libs_OneWire.html
> //
> // The DallasTemperature library can do all this work for you!
> // https://github.com/milesburton/Arduino-Temperature-Control-Library
>
> OneWire  ds(10);  // on pin 10 (a 4.7K resistor is necessary)
>
> void setup(void) {
>   Serial.begin(9600);
> }
>
> void loop(void) {
>   byte i;
>   byte present = 0;
>   byte type_s;
>   byte data[9];
>   byte addr[8];
>   float celsius, fahrenheit;
>
>   unsigned long initStartTime = micros(); // 记录初始化开始时间
>
>   if ( !ds.search(addr)) {
>     Serial.println("No more addresses.");
>     Serial.println();
>     ds.reset_search();
>     delay(250);
>     return;
>   }
>
>   unsigned long initEndTime = micros(); // 记录初始化结束时间
>
>   Serial.print("ROM =");
>   for( i = 0; i < 8; i++) {
>     Serial.write(' ');
>     Serial.print(addr[i], HEX);
>   }
>
>   if (OneWire::crc8(addr, 7) != addr[7]) {
>       Serial.println("CRC is not valid!");
>       return;
>   }
>   Serial.println();
>
>   // the first ROM byte indicates which chip
>   switch (addr[0]) {
>     case 0x10:
>       Serial.println("  Chip = DS18S20");  // or old DS1820
>       type_s = 1;
>       break;
>     case 0x28:
>       Serial.println("  Chip = DS18B20");
>       type_s = 0;
>       break;
>     case 0x22:
>       Serial.println("  Chip = DS1822");
>       type_s = 0;
>       break;
>     default:
>       Serial.println("Device is not a DS18x20 family device.");
>       return;
>   }
>
>   unsigned long waitStartTime = micros(); // 记录等待DS18B20释放总线开始时间
>
>   ds.reset();
>   ds.select(addr);
>   ds.write(0x44, 1);        // start conversion, with parasite power on at the end
>
>   //  delay(1000);     // maybe 750ms is enough, maybe not
>
> // Instead of delay
 while ( digital.Read(10) == 1 )  ;

>   // we might do a ds.depower() here, but the reset will take care of it.
>
>   unsigned long waitEndTime = micros(); // 记录等待DS18B20释放总线结束时间
>
>   present = ds.reset();
>   ds.select(addr);
>   ds.write(0xBE);         // Read Scratchpad
>
>   unsigned long readCmdStartTime = micros(); // 记录发送读数据命令开始时间
>
>
>
>

mayjack0312, 2024-06-22:
> El El sáb, 22 jun. 2024 a la(s) 11:27, LMY ***@***.***> escribió:
> #include <OneWire.h> // OneWire DS18S20, DS18B20, DS1822 Temperature Example // // http://www.pjrc.com/teensy/td_libs_OneWire.html [...]

I need a program to help me read the corresponding time in the table
[image]

uzi18, 2024-06-25:
This is because every time you use method with ending "byIndex" DallasTemperature library scans bus for your device
So device 0 is always first and second needs more time to communicate, as device 0 respond first and device 1 need to wait.
Solution is to scan bus first and store address of every sensor. 
Than use it later directly not by index.

mayjack0312, 2024-06-26:
> This is because every time you use method with ending "byIndex" DallasTemperature library scans bus for your device So device 0 is always first and second needs more time to communicate, as device 0 respond first and device 1 need to wait. Solution is to scan bus first and store address of every sensor. Than use it later directly not by index. By the way use only one code as base, they are almost similar .

Thank you, I'm trying it hard and feel like this is the final solution I want

closed 2024-06-26 by mayjack0312

uzi18, 2024-06-26:
Ok write down here if you found problem with this.