cpetrich/counterfeit_DS18B20 #40: The number of deviations is unexpectedly small.

dwrobel, 2024-08-14:
I found those sketches and decided to check my recent batch of DS18B20.

Using the latest version f37bd670aab9a81b1ecb83cdde7c083427c00d64 plus the fix for compilation errors:https://github.com/cpetrich/counterfeit_DS18B20/pull/39 I got the following output for 17 sensors connected.

One note that the first 10 I bought quite recently from a seller which claimed to sell only original Maxim sensors, the remaining 7 are quite old and I don't recall where I bought them:

classify_fake_DS18B20.ino output:
```
28-E4-FA-2F-57-23-0B-AF: Family B1 (Clone).
28-14-81-B9-57-23-0B-28: Family B1 (Clone).
28-B2-E6-3D-57-23-0B-51: Family B1 (Clone).
28-A1-93-E3-57-23-0B-F8: Family B1 (Clone).
28-35-DA-A8-57-23-0B-FB: Family B1 (Clone).
28-03-86-36-57-23-0B-E4: Family B1 (Clone).
28-F3-B0-B9-9A-23-0B-8E: Family B1 (Clone).
28-FB-DE-F1-57-23-0B-23: Family B1 (Clone).
28-17-DE-29-57-23-0B-6A: Family B1 (Clone).
28-8F-A6-2A-57-23-0B-A1: Family B1 (Clone).
28-FF-64-02-E2-7D-3D-B5: Family C (Clone).
28-FF-64-02-E2-63-DD-6C: Family C (Clone).
28-FF-64-02-E2-7F-EF-CF: Family C (Clone).
28-FF-64-02-E2-7F-3F-98: Family C (Clone).
28-FF-64-02-19-C8-AE-F7: Family C (Clone).
28-FF-64-02-19-86-B7-B1: Family C (Clone).
28-FF-64-02-19-81-16-2E: Family C (Clone).
```
discover_fake_DS18B20.ino output:
```
28-E4-FA-2F-57-23-0B-AF: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  Scratchpad Register: 50/05/4B/46/7F/FF/0C/10/1C
  Info only: Scratchpad bytes 2,3,4 (4B/46/7F):  Maxim default values.
  Scratchpad byte 5 (0xFF):  ok.
  Scratchpad byte 6 (0x0C):  ok.
  Scratchpad byte 7 (0x10):  ok.
  0x4E modifies alarm registers:  ok.
  0x4E accepts 10 bit resolution:  ok.
  0x4E preserves reserved bytes:  ok.
  0x4E accepts 12 bit resolution:  ok.
  0x4E preserves reserved bytes:  ok.
  Checking byte 6 upon temperature change:  ok.
  --> Sensor appears to be counterfeit based on 1 deviations.
  The number of deviations is unexpectedly small.
  Please see https://github.com/cpetrich/counterfeit_DS18B20/
  to help interpret the result.

28-14-81-B9-57-23-0B-28: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  Scratchpad Register: 50/05/4B/46/7F/FF/0C/10/1C
  Info only: Scratchpad bytes 2,3,4 (4B/46/7F):  Maxim default values.
  Scratchpad byte 5 (0xFF):  ok.
  Scratchpad byte 6 (0x0C):  ok.
  Scratchpad byte 7 (0x10):  ok.
  0x4E modifies alarm registers:  ok.
  0x4E accepts 10 bit resolution:  ok.
  0x4E preserves reserved bytes:  ok.
  0x4E accepts 12 bit resolution:  ok.
  0x4E preserves reserved bytes:  ok.
  Checking byte 6 upon temperature change:  ok.
  --> Sensor appears to be counterfeit based on 1 deviations.
  The number of deviations is unexpectedly small.
  Please see https://github.com/cpetrich/counterfeit_DS18B20/
  to help interpret the result.

28-B2-E6-3D-57-23-0B-51: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  Scratchpad Register: 50/05/4B/46/7F/FF/0C/10/1C
  Info only: Scratchpad bytes 2,3,4 (4B/46/7F):  Maxim default values.
  Scratchpad byte 5 (0xFF):  ok.
  Scratchpad byte 6 (0x0C):  ok.
  Scratchpad byte 7 (0x10):  ok.
  0x4E modifies alarm registers:  ok.
  0x4E accepts 10 bit resolution:  ok.
[... 252 lines cut ...]
28-F3-B0-B9-9A-23-0B-8E: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 3 deviations.
28-FB-DE-F1-57-23-0B-23: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 1 deviations.
28-17-DE-29-57-23-0B-6A: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 1 deviations.
28-8F-A6-2A-57-23-0B-A1: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 1 deviations.
28-FF-64-02-E2-7D-3D-B5: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 3 deviations.
28-FF-64-02-E2-63-DD-6C: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 3 deviations.
28-FF-64-02-E2-7F-EF-CF: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 3 deviations.
28-FF-64-02-E2-7F-3F-98: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 3 deviations.
28-FF-64-02-19-C8-AE-F7: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 3 deviations.
28-FF-64-02-19-86-B7-B1: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 3 deviations.
28-FF-64-02-19-81-16-2E: ROM does not follow expected pattern 28-xx-xx-xx-xx-00-00-crc. Error.
  --> Sensor appears to be counterfeit based on 3 deviations.
```

So the question is, are those first 10 really genuine or it is not possible to easily detect whether they are counterfeited?

cpetrich, 2024-08-14:
Thank you for sharing comprehensive data!

All of the sensors are counterfeit, I'm afraid: `classify_fake_DS18B20.ino` checks commands that genuine sensors ignore, so there is no doubt.

I came across several vendors stating that they sell authentic sensor when in fact they didn't. So, you're in good company.

Regarding the output from `discover_fake_DS18B20.ino`, I find it very interesting that the `28-xx-xx-xx-57-23-0B-crc` sensors appear to have a reworked scratchpad logic. I wonder if anything else has changed. I'll try to look into this. Thanks for sharing!

cpetrich, 2024-10-05:
Got a few samples of this variant and had a closer look: from what I can tell so far, under the hood they act just like Family B1 except that all aspects related to the scratchpad register have been modified to match the original sensors. I've split them off as a new Family B1-v2 in the document.
Again, I much appreciate you sharing this observation.

closed 2024-10-05 by cpetrich