cpetrich/counterfeit_DS18B20 #43: Updated sketches

divadiow, 2024-10-23:
Hi. Regarding

> (Note: as of 5 Oct 2024, the sketches are due for an update as they do not reflect the description of the Families A3, B1v2 and E-G on this page. Also, much of the content of this page does not yet account for those families.)

Are updated sketches imminent? :)

Thank you for all your efforts btw

cpetrich, 2024-10-23:
yes it is. I meant to upload a new sketch yesterday but then found another Family of clones while I tested that sketch against the chips in my drawer. It's in the works. Thank you for your interest.

divadiow, 2024-10-23:
awesome, cool.

I've got a few from Ali on order so am eager to check their status. 

I've currently a Sensylink - https://www.elektroda.com/rtvforum/viewtopic.php?p=21158431#21158431

and what's reporting to be a Maxim https://www.elektroda.com/rtvforum/viewtopic.php?p=21165112#21165112
though that's using https://github.com/koendv/CheckDS18B20/tree/main

divadiow, 2024-10-28:
well that was a bit boring. all 10 that arrived seem to be genuine. at least according to the current sketch

28-E1-43-5A-00-00-00-64: Family A1 (Genuine Maxim)
28-44-02-5B-00-00-00-78: Family A1 (Genuine Maxim)
28-B9-BD-56-00-00-00-FD: Family A1 (Genuine Maxim)
28-66-F1-58-00-00-00-C0: Family A1 (Genuine Maxim)
28-A6-B2-55-00-00-00-65: Family A1 (Genuine Maxim)
28-FD-AB-55-00-00-00-3C: Family A1 (Genuine Maxim)
28-F9-28-5A-00-00-00-FF: Family A1 (Genuine Maxim)
28-33-03-57-00-00-00-BA: Family A1 (Genuine Maxim)
28-CC-A3-58-00-00-00-45: Family A1 (Genuine Maxim)
28-5E-34-59-00-00-00-F3: Family A1 (Genuine Maxim)

Will the updated sketch include the naming for the Sensylink clones? eg

28-FF-64-1F-73-94-DB-60: Family C (Clone)

cpetrich, 2024-10-28:
Well, that looks actually like they are all clones of Family A3.
I think I'll play it save and stick to letter codes with the latest news on them in the repository.

cpetrich, 2024-11-10:
I've added an updated discovery and classification sketch for DS18B20 clones and removed the old sketches as they were obviously out of date. Additions to the new sketch are Families A3, B1v2, F, G, and H. (Tests for Family E had already been added to the old sketch earlier this year.)

closed 2024-11-10 by cpetrich