cpetrich/counterfeit_DS18B20 #48: wdt reset when uploading to esp8266 or esp32 from arduino cloud ide

austrisv, 2025-04-23:
Sorry if dumb question, probably something basic. Have not been using Arduino for ages. What could be wrong if all I'm getting in the serial monitor after upload is this:

```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
ets Jul 29 2019 12:21:46
```

From the output in arduino console it seems that all went well. esp32 itself is working fine, can upload esphome fw to it and operating as expected. Appreciate your help or ideas. 

```
Sketch uses 282997 bytes (21%) of program storage space. Maximum is 1310720 bytes.
Global variables use 21400 bytes (6%) of dynamic memory, leaving 306280 bytes for local variables. Maximum is 327680 bytes.

Flashing with command:/Users/austris/.arduino-create/esp32/esptool_py/4.5.1/esptool --chip esp32 --port /dev/cu.usbserial-2120 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 /var/folders/vv/ty3z2x7n4hl2c2_pc8184fqr0000gn/T/extrafiles1186943525/discover-classify_fake_DS18B20.bootloader.bin 0x8000 /var/folders/vv/ty3z2x7n4hl2c2_pc8184fqr0000gn/T/extrafiles1186943525/discover-classify_fake_DS18B20.partitions.bin 0xe000 /var/folders/vv/ty3z2x7n4hl2c2_pc8184fqr0000gn/T/extrafiles1186943525/tools/partitions/boot_app0.bin 0x10000 /var/folders/vv/ty3z2x7n4hl2c2_pc8184fqr0000gn/T/arduino-create-agent96983682/discover-classify_fake_DS18B20.bin
esptool.py v4.5.1
Serial port /dev/cu.usbserial-2120
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:22:d8:15:b8:50
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00055fff...
Compressed 17568 bytes to 12205...
Writing at 0x00001000... (100 %)
Wrote 17568 bytes (12205 compressed) at 0x00001000 in 1.5 seconds (effective 92.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 284.7 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 447.1 kbit/s)...
Hash of data verified.
Compressed 283360 bytes to 158404...
Writing at 0x00010000... (10 %)
Writing at 0x0001b5ba... (20 %)
Writing at 0x000245bd... (30 %)
Writing at 0x00029b59... (40 %)
Writing at 0x0002f014... (50 %)
Writing at 0x000343b1... (60 %)
Writing at 0x0003cc29... (70 %)
Writing at 0x00045468... (80 %)
Writing at 0x0004c07e... (90 %)
Writing at 0x00051639... (100 %)
Wrote 283360 bytes (158404 compressed) at 0x00010000 in 15.5 seconds (effective 146.7 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...
Ok
```

austrisv, 2025-04-26:
that's interesting, shame on me, I might have missed the pin business/did not think of checking them/changing for the board ... will give it a try. @krypton-x , you gave me hope! :))

lalo-uy, 2025-04-27:
You can use LED_BUILTIN to get the led in different boards.

#define pin_LED  LED_BUILTIN

austrisv, 2025-04-30:
LED_BUILTIN did not work for me, but setting correct onewire and led pin#'s solved the reset problem. Thank you for the hint!

closed 2025-04-30 by austrisv